brianc118/MPU9250 #1: Missing variable assignment in constructor

danterp, 2016-04-27:
MPU9250(long clock, uint8_t cs, uint8_t low_pass_filter = BITS_DLPF_CFG_188HZ, uint8_t low_pass_filter_acc = BITS_DLPF_CFG_188HZ){
        my_clock = clock;
        my_cs = cs;
        **my_low_pass_filter = low_pass_filter;
	my_low_pass_filter_acc = low_pass_filter_acc;**
    }

brianc118, 2016-05-01:
@danterp : can you please provide more info on the issue?

danterp, 2016-05-02:
Hi Brian!

The private variables **my_low_pass_filter** and **my_low_pass_filter_acc** are not assigned values anywhere in the header or cpp file, though they are used in the **init* function (on line 78 and 81 respectively).  I noticed, however, that the private variables **my_clock** and **my_cs** were assigned to the values (with very similar names) passed in from the constructor (seen in the header file).  My assumption then was that these private variables were intended to be assigned in the constructor.  My original post shows the code I added in bold.

brianc118, 2016-05-09:
Fixed

closed 2016-05-09 by brianc118