brianc118/MPU9250 #2: Magnetometer Doesn't Work (modified for Arduino 1.6.5)

danterp, 2016-05-09:
Hey Brian,

I made the few minor changes necessary in the sketch and libraries to allow the program to compile.  I'm able to get the gyro and accel data, but there seems to be an issue with the magnetometer as communication is never established.

Have you had any problems with the magnetometer?  Would you be able to update the code to 1.6.x? I switched over to use the RTIMUlib I2C code, which is working, but the problem I have is that my boards were already designed to use SPI (I didn't expect these issues) and I lose a lot of valuable pins by tying my SPI and I2C pins together with hookup wire (the easiest reversible hack).

brianc118, 2016-05-09:
Hi @danterp . I have always been using the Teensy 3.x for my work. Normal Arduinos won't compile since digitalwritefast is not built into the Arduino core. Replacing digitalWriteFast with digitalWrite should fix all compilation errors.

Could you please elaborate on how "communication is never established"?
You say that the gyro and accel work which means SPI communication is working fine.

brianc118, 2016-05-09:
Also I am using Arduino 1.6.3

danterp, 2016-05-10:
The Serial Monitor displays this when I run the example sketch (with the necessary mods):

> Successful connection
> Failed connection to mag: 0

Previously the magnetometer data (columns 7, 8, 9) have been all zeros.  However, for some reason, this time around I'm seeing data in those columns.

-0.05	0.05	-0.21	-0.01	0.00	     0.99	**-15.35	13.43	16.57**	30.26
-0.29	0.08	-0.20	-0.00	-0.00    0.99	**-16.52	13.13	16.87**	30.32

danterp, 2016-05-10:
It looks like this is valid data.  I took my (metal) wire cutters and noticed the number change.  Very odd.

closed 2016-05-15 by brianc118