brianc118/MPU9250 #3: setSCK function

ijunglee, 2016-05-15:
Hello, I am trying this code and could not use the setSCK function.
The error message shows
MPU9250.ino: In function 'void setup()': 
MPU9250:38: error: 'digitalWriteFast' was not declared in this scope
MPU9250:40: error: 'class SPIClass' has no member named 'setSCK'
'digitalWriteFast' was not declared in this scope
I think that the SPI.h is the problem.
Would you please provide me the SPI.h which you use?
I use Arduino 1.6.5.
Thanks in advance =)

brianc118, 2016-05-15:
Hi!
This was originally written for the Teensy 3.0/3.1/3.2. setSCK and digitalWriteFast are only available on the Teensy

I will fix this in the next 15 min or so, and will push to the repo.

brianc118, 2016-05-15:
Try it now. Should compile :)

To make it work you may need to change SPI_CLOCK to a clock that works for whichever Arduino you're using. In the example it's set at 8000000 = 8MHz

ijunglee, 2016-05-15:
Thanks for your reply. It works! =D

Furthermore, I would like to ask two more questions:

1. 
I try to use Arduino to connect to two MPU9250 via SPI.
And I define two variables for the each cable select of MPU9250:

#define SS_PIN   10
#define SS_PIN2   8

When I set the SS_PIN and SS_PIN2 to OUTPUT mode:

pinMode(SS_PIN, OUTPUT);
pinMode(SS_PIN2, OUTPUT);

The result is so weird. It shows:

[image]

The values of acceleration of Sensor 1 are wrong. 

But if I didn't set the SS_PIN and SS_PIN2 to OUTPUT mode, it shows:

[image]

The values of acceleration are all fine.
Why don't I need to set SS_PINs to OUTPUT mode?

2. 
Sometimes I can not get the values of magnetometer. It shows( 0, 0, 0 ). 
However, when I reset the arduino by disconnecting and re-connecting it, I can get the values well. (But I don't really understand about how to check the values of gyroscope and magnetometer.) 

Do you have any idea of the problem as mentioned above?
Thanks for your kindly help!
Hope to have discussion with you!

abstruse123, 2016-05-16:
Dear Mr. BrianC,

I am trying to port this code to PIC MCU.

SPI read operation does not return any data from the MPU 9250. 
SDO of MPU 9250 seems to always at zero. 

Would you mind to help?

brianc118, 2016-05-16:
@abstruse123 . Sorry are you really don't anything about PIC. I also don't understand your question. Also next time if it's something completely unrelated don't post under the same issue.

@ijunglee .
1. I'm not exactly sure, it may have something to do with how SPI slave select pins are configured
2. Yes there is probably a bug in the code somewhere. In the sample code it checks if the magnetometer is successfully connected at the start in setup(). If it isn't the program still runs but only reads the gyro and accelerometer. A somewhat sketchy fix would be to try connecting to the magnetometer multiple times instead of just once, to increase the chance of success.

closed 2016-05-16 by brianc118

abstruse123, 2016-05-16:
@brianc118 I am really Sorry.
I couldn't find any other helpful resource than yours on the internet.
I posted it here as I am new to github and didn't know how else to contact you.
You may delete this if you wish to.
Thanks.

brianc118, 2016-05-16:
haha I'm not too familiar with it myself! Create a new issue called "porting to PIC" or something similar and we'll start from there.