brianc118/MPU9250 #4: Magnetometer reading (0,0,0) - basically not connected

brianc118, 2016-05-16:
Known bug as mentioned #2 and #3. To be fixed when I have time

brianc118, 2017-02-18:
Fixed with #13

closed 2017-02-18 by brianc118

mattdabin, 2017-02-18:
Hi Brian,

Do you know the source of the bug?

Cheers,
Matt

brianc118, 2017-02-19:
Essentially to do with INT_PIN_CFG. Was set at 0x30 before, but apparently that doesn't work. 0x12 and 0x22 both work. See [this post](https://github.com/kriswiner/MPU-9250/issues/62#issuecomment-221782636).

MitalPattani, 2018-06-08:
Even this isn't working! i tried INT_PIN_CFG values to 0x12 as well as 0x22 still doesn't work.

brianc118, 2018-06-08:
@MitalPattani Interesting. Currently very busy until end of next week. I will try to play around with this and see what's going on then; it's been a while since I wired up the sensor with an arduino. Seems like some people are able to get the magnetometer working but some can't? In the meantime you might want to check out this [newer library](https://github.com/bolderflight/MPU9250) which looks promising; if it works please let me know.

MitalPattani, 2018-06-08:
Hey Brian, thanks, I tried the library still the same issue!