brianc118/MPU9250 #5: porting to PIC

abstruse123, 2016-05-17:
I attempted to port the code to PIC MCU.

Initially, I read the contents of MPU-9250's WHO AM I register and AK8963's WHO AM I register correctly i.e. 0x71 and 0x68 respectively. The contents of these registers are displayed on UART.

Then, I proceeded with reading all the registers of MPU-9250. I put everything in a while loop where in every cycle the WHO AM I register of MPU 9250 is read and only if its found to be 0x71, it proceeds with reading the registers.
In this case, only during the first cycle the WHO AM I registers are correctly read and readings are displayed correctly. However, in the subsequent cycles, the content of MPU-9250 WHO AM I register is displayed as 0x70 on UART and since it is not 0x71, the registers are not read and its coming out of the loop. 

When I see on the oscilloscope, the SDO from MPU-9250 is 0x71 only. However, it is displayed as 0x70 on UART.
Where could be problem? Pl. help

brianc118, 2016-05-17:
Just to clarify, WHO AM I is reading 0x70?

Are you running the sample code?

I2C addresses are always 7 bit so perhaps there was something wrong in the conversion that caused the last bit to be lost?

0x70 is 0111 0000
0x71 is 0111 0001

Sorry I can't quite help you here unless you help pinpoint the problem.

abstruse123, 2016-05-18:
@brianc118 Please understand that '0x71' is the device ID(data) stored in MPU-9250's address location '0x75' (WHO AM I register).
Moreover, I am not using I2C and moreover 0x70 is data read out. It is not the address.
Also, I could not find the '.ino' file in your project, where the functions in '.cpp' file are to be instantiated?

closed 2017-02-18 by brianc118