brianc118/MPU9250 #6: [Calibration]Gyro and Accelermeter offset

YanJenHuang, 2016-05-25:
Hi @brianc118 ,
I am new to the MPU9250, and I find that your code implements and comments are very easy to understand.
I have some questions about the MPU9250.cpp file. I just noticed that you store the gyro offset, and accel offset to the offset registers when you calibrate the MPU9250 by calibrate() function.
Since you had store the offset to the registers, why do you need to subtract the a_bias and g_bias in these two function again, MPU9250::read_acc(); MPU9250::read_gyro()? Wont it be subtracted by the MPU9250 itself?

brianc118, 2016-05-25:
Hi @YanJenHuang Huang you're probably right though i can't be sure until I check the  code. 

Unfortunately i don't have time until the weekend since i'm currently busy with uni work.

YanJenHuang, 2016-05-25:
I'll try to understand the implement of the MPU9250, initial, and SPI communication, etc. 
Now I am still looking at your code and check with the MPU9250 Register-Map! If you have any advice to me, I am willing to listen.
Thanks~

closed 2017-02-18 by brianc118