brianc118/MPU9250 #11: Disable I2C slave mode bus during initialisation

brixworth, 2016-12-01:
Hi Michael,

Just come across your driver for the MPU9250 (good work). As the driver code is using SPI to communicate with the device shouldn't the initialisation code disable the I2C interface?

Section 7.1 of MPU-9250 Product Specification states that the I2C_IF_DIS in the USER_CTL register should be set. If there is only one SPI device on the bus then no problem but if there is more than one and the I2C slave bus isn't disabled, then there is a danger of the device erroneously responding to a message that is not intended for it.

So, according to the documentation, should line 89 be changed from (this should not have any effect on the devices master mode I2C bus used to communicate with aux devices): 

        {0x20, MPUREG_USER_CTRL},       // I2C Master mode

To

        {0x30, MPUREG_USER_CTRL},       // I2C Master mode and set I2C_IF_DIS to disable slave mode I2C bus.

brianc118, 2016-12-01:
Hi @brixworth

I don't have an mpu-9250 close by me right now so won't be able to test, but it sounds right. I will make the changes soon.

Brian

brixworth, 2016-12-01:
Thanks Brian!

Cheers,

Mark.

closed 2016-12-01 by brianc118