brianc118/MPU9250 #12: Line 118 MPU9250.cpp

radioelf, 2017-01-17:
Hi.
Line 118  "//calib_mag();  // Can't load this function here , strange problem?"
I think it is necessary to run in the function calib_mag():

  WriteReg(AK8963_CNTL1, 0x00); // set AK8963 to Power Down
  delay(50); 
  WriteReg(AK8963_CNTL1, 0x0F); // set AK8963 to FUSE ROM access

...and

    WriteReg(AK8963_CNTL1, 0x00); // set AK8963 to Power Down
    delay(50);
    // Set magnetometer data resolution and sample ODR
    WriteReg(AK8963_CNTL1, MFS_16BITS << 4 | M_8HZ);
    delay(10);

brianc118, 2017-01-18:
Hi @radioelf . I'm not sure what you're saying here (specifically where should these snippets go?). Can you please send your version of MPU9250.cpp over so I can see the differences?

radioelf, 2017-01-18:
Hi. 
My code:
void MPU9250::calib_mag(){
    uint8_t response[3];
    float data;
    int i;
    // Choose either 14-bit or 16-bit magnetometer resolution
    //uint8_t MFS_14BITS = 0; 				                    // 0.6 mG per LSB
    uint8_t MFS_16BITS =1;  				                    // 0.15 mG per LSB 
    // 2 for 8 Hz, 6 for 100 Hz continuous magnetometer data read  
    uint8_t M_8HZ = 0x02;   				                    // 8 Hz update
    //uint8_t M_100HZ = 0x06;  				            // 100 Hz continuous magnetometer

    /* get the magnetometer calibration */

    WriteReg(MPUREG_I2C_SLV0_ADDR,AK8963_I2C_ADDR|READ_FLAG); 	// Set the I2C slave    addres of AK8963 and set for read.
    WriteReg(MPUREG_I2C_SLV0_REG, AK8963_ASAX); 				// I2C slave 0 register address from where to begin data transfer
    WriteReg(MPUREG_I2C_SLV0_CTRL, 0x83); 						// Read 3 bytes from the magnetometer

    //WriteReg(MPUREG_I2C_SLV0_CTRL, 0x81);    					// Enable I2C and set bytes
    delayMicroseconds(10000);  
    //response[0]=WriteReg(MPUREG_EXT_SENS_DATA_01|READ_FLAG, 0x00); //Read I2C 
 
    WriteReg(AK8963_CNTL1, 0x00);                               // set AK8963 to Power Down
    delay(50);                                                  // long wait between AK8963 mode changes
    WriteReg(AK8963_CNTL1, 0x0F);                               // set AK8963 to FUSE ROM access
    delay(50);                                                  // long wait between AK8963 mode changes

    ReadRegs(MPUREG_EXT_SENS_DATA_00,response,3);
    //response=WriteReg(MPUREG_I2C_SLV0_DO, 0x00);    			// Read I2C 
    for(i = 0; i < 3; i++) {
        data=response[i];
        Magnetometer_ASA[i] = ((data-128)/256+1)*Magnetometer_Sensitivity_Scale_Factor;
    }
    WriteReg(AK8963_CNTL1, 0x00); // set AK8963 to Power Down
    delay(50);
    // Configure the magnetometer for continuous read and highest resolution.
    // Set bit 4 to 1 (0) to enable 16 (14) bit resolution in CNTL
    // register, and enable continuous mode data acquisition (bits [3:0]),
    // 0010 for 8 Hz and 0110 for 100 Hz sample rates.   
    WriteReg(AK8963_CNTL1, MFS_16BITS << 4 | M_8HZ);            // Set magnetometer data resolution and sample ODR
    delay(10);
}

brianc118, 2017-01-19:
I've added this in now.

closed 2017-01-19 by brianc118