brianc118/MPU9250 #15: add filtering

aster94, 2017-10-10:
hello Brian,

your lib is the first one where the magnetometer works. my congratulation
why don t you add a filtering?
examples:
https://github.com/kriswiner/MPU9250/blob/master/quaternionFilters.ino
or the improved sparkfun version:
https://github.com/sparkfun/SparkFun_MPU-9250_Breakout_Arduino_Library

aster94, 2017-10-10:
ah anyway it seems that new mpu have different values of the who i am register, this is what my mpu9250 output
MPU: 0x73
MAG: 0xB6

brianc118, 2017-10-10:
The point of the library is to give you the raw sensor outputs. You can easily just take an off the shelf filter and apply it without any difficulty. I think it'd only make sense to "add filtering" if we were using the DMP which does it for you ... but we aren't using the DMP.

Beware, quarternion/mahony filters are kind of a black box. If you don't strictly need them I recommend you just use something more simple like a complementary filter.

brianc118, 2017-10-10:
Also the mag on the MPU-9250 really isn't that great. As a result, often it can mess up your filter results if you put too much weight on it.

aster94, 2017-10-10:
Ok brian thank you anyway, i understand the point of your library
My need was to have yaw, pitch and roll but the math behind them is too difficult for me so i will need to use an already made library for them

closed 2017-10-14 by brianc118