brianc118/MPU9250 #18: Error when compiling Basic_SPI.ino / Arduino mega SDK

Manujaya94, 2018-01-16:
I've got an error when compiling the Basic_SPI.ino to Arduino Mega ADK board. 
"
Build options changed, rebuilding all
Archiving built core (caching) in: /tmp/arduino_cache_254679/core/core_arduino_avr_megaADK_a9e1fb321504b6baabee80ff3578a86c.a
/tmp/ccJ37iTP.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_Basic_SPI.ino.cpp.o.2181':
ccJ37iTP.ltrans0.o:(.text.startup+0x7a): undefined reference to `MPU9250::MPU9250(SPIClass&, unsigned char)'
/tmp/ccJ37iTP.ltrans0.ltrans.o: In function `main':
ccJ37iTP.ltrans0.o:(.text.startup+0x1cc): undefined reference to `MPU9250::begin()'
ccJ37iTP.ltrans0.o:(.text.startup+0x272): undefined reference to `MPU9250::readSensor()'
ccJ37iTP.ltrans0.o:(.text.startup+0x27a): undefined reference to `MPU9250::getAccelX_mss()'
ccJ37iTP.ltrans0.o:(.text.startup+0x28a): undefined reference to `MPU9250::getAccelY_mss()'
ccJ37iTP.ltrans0.o:(.text.startup+0x29a): undefined reference to `MPU9250::getAccelZ_mss()'
ccJ37iTP.ltrans0.o:(.text.startup+0x2aa): undefined reference to `MPU9250::getGyroX_rads()'
ccJ37iTP.ltrans0.o:(.text.startup+0x2ba): undefined reference to `MPU9250::getGyroY_rads()'
ccJ37iTP.ltrans0.o:(.text.startup+0x2ca): undefined reference to `MPU9250::getGyroZ_rads()'
ccJ37iTP.ltrans0.o:(.text.startup+0x2da): undefined reference to `MPU9250::getMagX_uT()'
ccJ37iTP.ltrans0.o:(.text.startup+0x2ea): undefined reference to `MPU9250::getMagY_uT()'
ccJ37iTP.ltrans0.o:(.text.startup+0x2fa): undefined reference to `MPU9250::getMagZ_uT()'
ccJ37iTP.ltrans0.o:(.text.startup+0x30a): undefined reference to `MPU9250::getTemperature_C()'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Mega ADK.
Invalid library found in /home/manujaya/Arduino/libraries/Basic_SPI: /home/manujaya/Arduino/libraries/Basic_SPI
"

brianc118, 2018-01-16:
More info would be useful. What is "Basic_SPI"? Is that your program sketch?
Have you included MPU9250.h at the top?
Are you using the Arduino IDE?

It appears you are using functions that don't exist in the library such as getAccelY_mss().
I have no clue what this is but sounds like you don't really know what you're doing. Unfortunately I can't help you much with that. Start from the example sketches and work your way from there. Issues here should mainly be about bugs/improvements etc :)

closed 2018-01-16 by brianc118

Manujaya94, 2018-01-16:
Actually i am sorry for the above error . That was actually my mistake.