brianc118/MPU9250 #20: MISO is not working with TinyTile and MPU9250

GeorgeMcAlistar, 2018-01-25:
Hey, I'm refering to your code _'Sample program for the MPU9250 using SPI'_ of **MPU-9250 SPI Library for Arduino/Teensy**. 
I'm using a TinyTile to communicate with the MPU9250 (Break-Out). I tried to work it with I2C previously, which worked - so the MPU is working/ is undamaged. In order to combine this device with another device (which is using SPI) I like to use SPI and tried your code to work it out. 
Unfortunately I couldn't establish a proper connection/data exchange. As a result for the WhoAmI registers I always receive FF or 00. When I checked the signal via Logic analyzer there were signals for CLK, CS (SS, NSS) and MOSI but nothing for MISO - that might explain, why I didn't get a proper result for my WhoAmI requests, but I don't get, why the MPU is not responding. 

My Settings:
TinyTile----------------------------MPU9250
PIN 13                     -->                          SCL
PIN 12 (MISO)         -->                         SDO
PIN 11 (MOSI)         -->                         SDA
PIN 10 (CS)              -->                           CS
3.3V                         -->                        VDD
GND                        -->                        GND

SPISettings(my_clock, MSBFIRST, SPI_MODE3) with my_clock = 1000000  // 1MHz

#define SS_PIN   10 
#define INT_PIN  2
#define LED      13               (I'don't know, if I need that ...)
pinMode(INT_PIN, INPUT);
pinMode(SS_PIN, OUTPUT);
pinMode(LED, OUTPUT);
digitalWrite(LED, HIGH);

That's it. I hope someone can come up with an idea. 

PS: I checked the connections and cables, they are working.

brianc118, 2018-01-25:
@GeorgeMcAlistar So you are getting nothing out of MOSI/SDA? Your wiring looks correct. Don't be concerned if whami is returning 0xFF or 0x00 (I can't remember if it ever worked :/). The key is whether the MPU9250 is sending back data from the gyroscope/accelerometers.

>  In order to combine this device with another device (which is using SPI) I like to use SPI and tried your code to work it out.

What do you mean here? Is there another SPI device connected to the TinyTile? If so remove it first and test.

Unfortunately it's a bit hard for me to help here because the problem is probably not obvious

brianc118, 2018-01-25:
Also I'm not familiar with the TinyTile. Have you successfully got any other SPI devices working with it? There doesn't seem to be as much support from google.

GeorgeMcAlistar, 2018-01-25:
Hey Brian, thanks a lot for your reply.
I'm getting data from the MOSI/SDA,  as it's the output of my TinyTile. But
there"s nothing on the MISO, as if the MPU9250 isn't replying...
I checked the channels with Logic Analyzer (see attachment). The TinyTile
is basically a smaller Version of the Arduino UNO, many things are quiet
similar. What Board did you use to communicate with the MPU9250? And how
did you wire it?

I plan to combine the MPU with another SPI device on the tinytile, but
right now I'm just working with the MPU and the TinyTile.
I run the programm just again, as I wrote the mail and guess what, I've got
something from the MISO but it's not changing

GeorgeMcAlistar, 2018-01-25:
yeah, doesn't seem like a proper reply

GeorgeMcAlistar, 2018-01-30:
Hey Brian,
I got it working so far. I receive a MISO signal. Problem was the solder
jumper 2, which bypassed the output line of my device.
I receive various data for the gyroscope and accelerometer - that's fine,
but I can't access the the magnetometer. As you put a questionmark behind
the magnetometer
on your GitHub entry (Invensense MPU-9250 SPI Library)  I guess, you had
some trouble, too.
What was the issue and how did you solve it?

Cheers,
Marco

closed 2019-04-25 by brianc118