brianc118/MPU9250 #21: Miso Mosi pins not defined in library

Devanshi15, 2018-06-01:
I was using this library for Nordic semiconductor board Redbearlab nrf51822 board in arduino IDE PCA10028 board .Example code got compiled but I see garbage values in serial monitor. And failed connection . I had defined pins like as follows

#define SPI_CLOCK 8000000  // 1MHz
#define SPI_CLOCK 16    /////P0.25 
#define CS   10    ////P0.14
#define MISO    17    /// P0.22 SDO
#define MOSI    18     ////P0.20  SDI
#define LED      13  ////P0.15
#define INT      12 ///P0.13

I need to know that do there is any support to change the default pin it is taking for spi communication. And where do I change it...???
Thank you....

brianc118, 2018-06-03:
@Devanshi15 So you are looking to connect the MPU9250 with your PCA10028? Provided the Arduino Code has support for SPI on PCA10028, it should work. SPI pins to a certain pins on any microcontroller and it is not a matter of just changing it? In fact, chances are there's only one SPI port on your board meaning you can't change it anyway.

The code snippet you provided does not give me any context to understand the problem. What does defining MISO/MOSI achieve? Where is it used in your code? It certainly isn't used in the example provided.

This library does not provide direct support changing the default pin for SPI communication. That would be up to you to change in your code. For example, if you were using a Teensy instead of PCA10028 you could use SPI.setMISO and SPI.setMOSI to change the port.

brianc118, 2018-06-04:
The Arduino SPI library uses the default port. This library uses the core arduino SPI library, which should use whatever is the default SPI port as specified by your board, provided your board has an SPI port. It should be a matter of searching up "nrf51822 pinout" and looking for the MISO/MOSI pins and connecting MISO (dev board) - MOSI (mpu9250) and MOSI (dev board) - MISO (mpu9250).

> It is working fine when I try to serially communicate something. So, it should support this library too.
I'm not sure what you mean by this. What do you mean by "Serial" communication? As in via UART? That is completely irrelevant.

closed 2018-06-05 by brianc118